uf-mil/SubjuGator #89: Get a PPA working (Or some solution for distributing binaries)

jpanikulam, 2016-04-05:
* In the short term, everyone is going to have to compile PCL from source

jpanikulam, 2016-04-12:
Interim solution: Going to make everyone compile pcl from source.

whispercoros, 2016-05-19:
I will be trying to download, build, package, and distribute the following in the next few weeks:

* OMPL (current)
* PCL (1.8)
* ArrayFire (current)
* OpenCV (2.4.13 -cuda -opencl -mp)
* Ceres (current)
* gpu-voxels (current)

If anyone has any additions or suggestions as to how to best build these, please comment below. I will be building for amd64, i386, and arm (where possible) platforms.

whispercoros, 2017-01-31:
Since the move to Ubuntu 16.04 and ROS Kinetic, I think all of the dependency version issues we were having have been resolved. Can anyone confirm whether or not we still need any of the binaries listed above to be custom compiled?

whispercoros, 2017-02-07:
Closing this due to lack of activity.

closed 2017-02-08 by DSsoto