uf-mil/SubjuGator #98: Add area search to channel marker mission segments

jpanikulam, 2016-04-12:
* Have some knowledge of where we should look and where we have already looked
* Visualize it in Rviz

mattlangford, 2016-05-07:
Can this be closed or should we wait until it has been tested irl

jpanikulam, 2016-05-07:
I am sufficiently satisfied that this is solved

jpanikulam, 2016-05-07:
What'll be your strategy for making sure the search pattern doesn't hit the pool walls?

closed 2016-05-07 by jpanikulam

mattlangford, 2016-05-07:
So the sub won't search outside the occupancy grid and since we know approximately where we start in the transdec (right?) we can define the grid such that the sub won't hit the walls.
Also for testing in pools we can define the grid to be approx the size of the pool based on where we put the sub in the water.

jpanikulam, 2016-05-07:
Ok.

One of these days I'd like to have a single place (maybe a launch arg or param) that configures the sub to load calibrations for one of [gazebo, transdec, Town Parc pool]

Then, most importantly, lets the operator align a model of the pool in RVIZ.

mattlangford, 2016-05-07:
Yeah that sounds like a good idea. 
Doing the pool alignment could be tricky, what were you thinking for that? If we made 3D model of the pool, positioning it would be tricky since we still wouldn't know where the sub was in the pool initially. Would that just be an approximate representation of the pool, if so what would that do for us besides being a starting point for the marker occupancy grid?