uf-mil/SubjuGator #113: Add mission timeouts

jpanikulam, 2016-04-25:


mattlangford, 2016-05-07:
I can include this in my next PR

jpanikulam, 2016-05-07:
Perfect

jpanikulam, 2016-05-17:
@mattlangford Did we ever land on this?

mattlangford, 2016-05-17:
No reply from @forrestv yet. What do?

jpanikulam, 2016-05-17:
Have you tried emailing god-president Trump?

jpanikulam, 2016-05-17:
I couldn't find the timeout code on your fork, did you remove it? I was going to clone and try

mattlangford, 2016-05-17:
I did remove it. I can add it back tonight.

forrestv, 2016-05-18:
>No reply from @forrestv yet. What do?

Huh? Not sure what you mean. Are you waiting on me fixing the spurious traceback that is printed in the case you emailed me? I thought it wasn't blocking any progress..?

mattlangford, 2016-05-18:
Hmm I guess you're right. It doesn't actually break anything, no reason to fret over it.

forrestv, 2016-05-18:
Well, I'm definitely going to look into it. I just wanted to point out that
it shouldn't prevent you from advancing.

I'm still a bit confused, though: that issue you showed me, I wonder why I
never saw it before when I was writing missions... From the script you
showed me, it didn't look like you were doing anything unusual... I have to
try to replicate it locally.

FYI, also, I'm likely going to redo the txros action API. It's a mess right
now.

mattlangford, 2016-05-18:
@jpanikulam, have you tried running it yet? I wonder if it's just a weird issue on my machine. I haven't ran it anywhere else.

Happy birthday by the way Forrest

ironmig, 2019-01-08:
Will be done via #320

closed 2019-01-08 by ironmig