uf-mil/SubjuGator #140: Make one-click simulated missions

jpanikulam, 2016-05-18:
* Take less than 5 mins to set up a gazebo test case
* Take less than 30 secs and 10 brain-cells (no cocaine) to start running an already existing test case

jpanikulam, 2016-05-18:
As in Semaphore, zero effort is expended to see if
* install script runs on a fresh install
* flake8 passes
* build works
* unit-tests and integration tests run

We should be able to do something similar in Gazebo. Being as painless as Semaphore is probably aiming too high, but minor improvements over @mattlangford 's current JobRunner implementation will save everybody days of test time.

jpanikulam, 2016-05-18:
What I think this would look like:

* CLI like missions
```shell
# Internally runs/stops/restarts the appropriate mission as a subprocess
rosrun sub8_gazebo test test_gazebo
```

But I am just now trying to write a Gazebo job using @mattlangford 's script and finding it to be even easier than I thought

mattlangford, 2016-05-18:
So my idea was that the gazebo server would constantly be running the jub_runner, then people could start up missions anywhere on the network. That being said I'm sure it wouldn't be difficult to modify the script to be all in one and that would make it easier to use and would save dev time. The main thing my script would need is some identification over if the mission passed or failed (for bagging and statistics) which could be refactored by just returning information from the mission script.

whispercoros, 2016-05-19:
@mattlangford, what happened to the job_runner script? I am going to try to create a server on the gazebo machine to do 24/7 testing. Basically, there would be a web interface (or maybe just an SSH terminal interface if that takes too long) where jobs can be submitted, monitored, and controlled. A master queue would run the following in the specified order of priority:

* A specific set of tests required for every version of the master branch and pull requests (so we can compare the accuracy of each branch)
* Jobs submitted by individual users to be run a set number of times
* A series of tests that are to be cycled through when nothing else is being run

Basically, this will ensure there are no gaps in our software and provide us with cool statistics (@jpanikulam really wants a mileage ticker). Sound good? Any suggestions?

mattlangford, 2016-05-19:
It's in sub8_gazebo/diagnostics. I think a web interface would be cool.
We do have a mileage ticker in the gazebo controller.

mattlangford, 2016-05-19:
It might be easier to rosify the mission adding process and make an rqt plugin that lets you monitor mission status.

whispercoros, 2016-05-19:
A git-independant mileage counter would be a lot simpler to manage and fairly easy to set up. If I implement this, we should be able to send all kinds of statistics to be centrally tracked from any Gazeb session.

Rosifying the process might be easier, but it'd be difficult to get working remotely. Ros really likes to use all kinds of crazy ports, so I'd prefer to make something ros-independant to submit jobs. My assumption is that jobs created with job_runner are a single script, so an upload or reference to a committed script would be much cleaner.

mattlangford, 2016-05-19:
Ah, I see what you're saying remote would be cool. 

Yeah you'd upload one or two scripts to run with it. Normally I'd have the mission script that would run on the sub then a mission running script that handles any errors or whatnot and checks for success or failure.
It could be one script though.

DSsoto, 2017-01-15:
Anthony might expand on this idea later on but I'll this seems resolved.

closed 2017-01-15 by DSsoto