uf-mil/SubjuGator #155: ML: Store "observe" function/kernels along with classifiers

jpanikulam, 2016-06-08:
Right now, if someone modifies features.py after a classifier is trained, the classifier simply won't work anymore

mattlangford, 2016-06-08:
You right, you think just appending the list of features to the classifer pickle will work?

jpanikulam, 2016-06-08:
I think so. ought to make a unified `observe` to make sure there's no goof-ups. 

Also, Boosting can tell you which kernels are being useful, so you could save convolutions by selectively using only the ones that help

mattlangford, 2016-06-08:
Did you want to do that? I can possibly tonight after I do some testing depending on how late it is when I finish what I'm working on rn

mattlangford, 2016-06-08:
Also do you think it might be good to save the cv2 classifiers in a pickle as well vs using an xml which it does now? That would make switching between the two really simple. Esp if we wanting to include other things in the pickle like kernels

jpanikulam, 2016-06-08:
All you, my friend. 

Unfortunately, cv2 classifiers can't be pickled, because they are made as Boost::Python bindings, so we have to use their serialization interface

mattlangford, 2016-06-08:
Ohh the data doesn't exist in the python world.
I guess you could pickle the xml somehow

jpanikulam, 2016-06-08:
I'm not sure it would be worth the investment of effort, but...
I think you could come up with a trick where you write it to /dev/stdin, [grab it](http://stackoverflow.com/questions/1450393/how-do-you-read-from-stdin-in-python), then pickle that...and then when you read it, you read from /dev/stdout

or something ridiculous like that

mattlangford, 2016-06-08:
I'm just trying to figure out a good way to save an arbitrary learning object that can be used by this theoretical interface.

closed 2017-01-15 by DSsoto