uf-mil/SubjuGator #160: odom_estimator consuming entire CPU core

rleyva, 2016-06-20:
We've noticed this behavior when pool testing and believe it's playing a part in unstable sub behavior. Below are some notes regarding it:
@jpanikulam - Mentioned that typical behavior has the sub running this node with ~15% usage

The setup that I am using to debug this is the following:
- Running fake_imu tool in the odom_estimator package under raw_gps
- rosrun tf static_transform_publisher 0 0 0.5 0 0 0 /map /base_link 10 (Random, needed to link tf trees together)
- rosrun tf static_transform_publisher 0 0 1 0 0 0 /base_link /imu 10

With this setup, the odom_estimator is in fact railing the CPU  core

forrestv, 2016-06-20:
Are you compiling with optimization enabled? Math-heavy code like
odom_estimator and c3_trajectory_generator that uses Eigen for
vector/matrix math incurs around a 1000% penalty when optimization is
disabled.

forrestv, 2016-06-20:
By optimization, I mean -O2 or -O3. You can check by running catkin_make
with --make-args "VERBOSE=1" added and looking at the g++ commands when it
gets to odom_estimator.

rleyva, 2016-06-21:
Whoa, I didn't realize that this was the case. We recompiled C3 with optimizations at the pool on Saturday but never bothered to recompile the entire workspace with optimizations. Recompiling odom estimator on my machine made the issue go away. I'll close the issue once we retest on Wednesday.

rleyva, 2016-06-22:
That did it. Thanks, Forrest!

closed 2016-06-22 by rleyva