uf-mil/SubjuGator #166: PERCEPTION: nodes crashing on startup

DSsoto, 2016-07-01:
When launching the sub via sub8.launch or the sim via duck.launch, the nodes follow _orange_pipes.py and buoy_3d.py consistently crash. These are the error messages:

follow _orange_pipes.py

> Exception in thread Thread-19:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/timer.py", line 223, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/home/santiago/sub_ws/src/Sub8/gnc/sub8_perception/nodes/follow_orange_pipes.py", line 41, in publish_target_info
    markers = self.find_pipe_new(np.copy(self.last_image))
  File "/home/santiago/sub_ws/src/Sub8/gnc/sub8_perception/nodes/follow_orange_pipes.py", line 122, in find_pipe_new
    xy_position, height = self.occ_grid.get_tf(timestamp=last_image_timestamp)
  File "/home/santiago/sub_ws/src/Sub8/gnc/sub8_perception/sub8_vision_tools/marker_occ_grid.py", line 295, in get_tf
    self.tf_listener.waitForTransform("/ground", "/downward", timestamp, rospy.Duration(1.0))
Exception: Lookup would require extrapolation into the past.  Requested time 1.301000000 but the earliest data is at time 2.038000000, when looking up transform from frame [downward] to frame [ground]. canTransform returned after 1.00755 timeout was 1.

buoy_2d.py

> OpenCV Error: Unspecified error (The boosted tree ensemble has not been trained yet) in predict, file /build/buildd/opencv-2.4.8+dfsg1/modules/ml/src/boost.cpp, line 1614
Exception in thread Thread-28:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/timer.py", line 223, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/home/santiago/sub_ws/src/Sub8/gnc/sub8_perception/nodes/buoy_2d.py", line 136, in publish_target_info
    self.find_buoys(np.copy(self.last_image))
  File "/home/santiago/sub_ws/src/Sub8/gnc/sub8_perception/nodes/buoy_2d.py", line 238, in find_buoys
    result = self.find_single_buoy(img, buoy_name)
  File "/home/santiago/sub_ws/src/Sub8/gnc/sub8_perception/nodes/buoy_2d.py", line 193, in find_single_buoy
    prediction2 = [int(x) for x in [self.boost.predict(obs) for obs in some_observations]]
error: /build/buildd/opencv-2.4.8+dfsg1/modules/ml/src/boost.cpp:1614: error: (-2) The boosted tree ensemble has not been trained yet in function predict

@mattlangford, @jpanikulam you guys might have some insight into these, I see the first one is tf related.

mattlangford, 2016-07-01:
The second error is due to the fact that the classifications for buoys aren't on the git and therefore aren't on your computer. So when it tries to load them it fails and dies.
The first seems odd, maybe it's starting up before tf has started - it would be good to waitForTransform at the begginning so it doesn't start until it sees the frames

DSsoto, 2016-07-01:
Why don't we put them on the git? Are they too large?

And yea, that would probably fix it.

mattlangford, 2016-07-01:
I think they are quite large. We should keep them somewhere, on the server maybe?

whispercoros, 2016-07-02:
I could have the sub automatically sync that directory just like the bags if you tell me where it is.

mattlangford, 2016-07-02:
Yeah that's what I was thinking. I guess they can go wherever ..../sub8_perception/ somewhere

mattlangford, 2016-07-02:
If things go right today, I'm going to be doing some segmentation tonight, so if you're in lab at all today @whispercoros, we can talk about it.

khaledhassan, 2016-07-03:
Perhaps consider Git LFS? Never used it, just a thought

mattlangford, 2016-07-03:
Jake had talked about using that for the gazebo models - we will look into it.

closed 2017-01-02 by mattlangford