uf-mil/SubjuGator #191: Proposed Wiki Changes

whispercoros, 2017-01-17:
Since we are moving toward development based on softwar-common, I thought it would be a good idea to move the wiki pages that apply to all projects or MIL in general over to that repository. On top of that, there were a few pages that I found that had no useful purpose and a few that need to be updated to make sense. If nobody objects, I will start executing the changes in a week. Feel free to add to, remove from, or change the list in this issue.

I do not have the access rights to delete wiki pages, so someone with access will need to do that part. Items with a * in the comment after them will need to be taken care of by someone actually working on that project, as I so not have the knowledge to update them.

Pages that serve no purpose and should be removed:
* [General To Do (Mechanical)](https://github.com/uf-mil/Sub8/wiki/General-To-Do-(Mechanical)) (According to the great Ken, himself)
* [Goal Manager](https://github.com/uf-mil/Sub8/wiki/Goal-Manager) (Seems to be based on an outdated NaviGator package)
* [Model Learning](https://github.com/uf-mil/Sub8/wiki/Model-Learning) (This page is very sparse and seems to be targeted twoard someone who no longer works in MIL)
* [Networking](https://github.com/uf-mil/Sub8/wiki/Networking) (No longer needed, networking pages are being updated and moved to the software-common repository)
* [Sonar Box To Do](https://github.com/uf-mil/Sub8/wiki/Sonar-Box-To-Do-List) (Again, according to Genghis Ken)
* [Sonar Box To Do List](https://github.com/uf-mil/Sub8/wiki/Sonar-Box-To-Do-List)

Pages that are outdated and will only make sense if updated:
* [Implementation Plans](https://github.com/uf-mil/Sub8/wiki/Implementation-Plans) (* Not current, should be updated before we get too far into this development cycle)
* [Mission Scheduler](https://github.com/uf-mil/Sub8/wiki/Mission-Scheduler) (Maybe hide this for history? The Git project for this seems to be more up to date)
* [Open Questions](https://github.com/uf-mil/Sub8/wiki/Open-Questions)
* [Perception Stack](https://github.com/uf-mil/Sub8/wiki/Perception-Stack) (*Outdated list of people working on tasks)
* [Pool Days](https://github.com/uf-mil/Sub8/wiki/Pool-Days) (Some of the commands need to be updated)
* [Visual SLAM](https://github.com/uf-mil/Sub8/wiki/Visual-SLAM) (*People working on tasks should be updated)

Pages that should be moved to software-common:
* [Accessing Bag Files](https://github.com/uf-mil/Sub8/wiki/Accessing-Bag-Files)
* [Aws](https://github.com/uf-mil/Sub8/wiki/Aws) (As a hidden page; our in-house simulation machines do the job well and are free)
* [C++ Spells](https://github.com/uf-mil/Sub8/wiki/C---Spells) (Change name [sorry Jake! and add a few lines to do the same thing in Python)
* [Code Policy](https://github.com/uf-mil/Sub8/wiki/Code-Policy)
* [Continuous Integration](https://github.com/uf-mil/Sub8/wiki/Continuous-Integration) (I will be expanding this to be more informative)
* [Editing the Wiki](https://github.com/uf-mil/Sub8/wiki/Editing-the-Wiki) (Or possibly have a copy on all Wikis?)
* [Getting Started](https://github.com/uf-mil/Sub8/wiki/Getting-Started) (All project readme files will link to this page)
* [Git](https://github.com/uf-mil/Sub8/wiki/Git)
* [Glossary](https://github.com/uf-mil/Sub8/wiki/Glossary)
* [History](https://github.com/uf-mil/Sub8/wiki/History) (Unlisted of course, and add a note about the MIL Renaissance)
* [Machine Learning Tools](https://github.com/uf-mil/Sub8/wiki/Machine-Learning-Tools)
* [Manifesto](https://github.com/uf-mil/Sub8/wiki/Manifesto)
* [MIL Local Network](https://github.com/uf-mil/Sub8/wiki/MIL-Local-Network) (Update to cover the network services running on vehicles)
* [Onboard Utilities](https://github.com/uf-mil/Sub8/wiki/Onboard-Utilities) (This will be rolled into the SSH Into Vehicles page)
* [Other Conventions](https://github.com/uf-mil/Sub8/wiki/Other-Conventions)
* [Reading List](https://github.com/uf-mil/Sub8/wiki/Reading-List)
* [Remote Access](https://github.com/uf-mil/Sub8/wiki/Remote-Access) (Needs to be updated a bunch, but will be moved)
* [ROS Conventions](https://github.com/uf-mil/Sub8/wiki/ROS-Conventions)
* [ROS things](https://github.com/uf-mil/Sub8/wiki/ROS-things)
* [SSH into the sub](https://github.com/uf-mil/Sub8/wiki/SSH-into-the-sub) (Needs significant updating)
* [SSH Keys](https://github.com/uf-mil/Sub8/wiki/SSH-Keys)
* [Unsupported OSes](https://github.com/uf-mil/Sub8/wiki/Unsupported-OSes) (This will need to be updated a bit)

duggerd, 2017-01-31:
I got rid of the following (any useful information will be migrated to Confluence):

- To Do (Electrical)
- Navigation Vessel (Electrical)
- Vessel Pinouts (Electrical)
- Spares (Electrical)
- Testing SW Stuff (Electrical)
- Sub Switch
- Sub Motherboard

The following have outdated information and should be updated/removed:

- Remote Access
- MIL Local Network

whispercoros, 2017-01-31:
Everything removed is okay. I am planning on converting [MIL Local Network](https://github.com/uf-mil/Sub8/wiki/MIL-Local-Network) to a page about the network on the actual vehicles (since they are almost identical) and [Remote Access](https://github.com/uf-mil/Sub8/wiki/Remote-Access) to a page about using UF's VPN to access the fileserver on Ubuntu.

duggerd, 2017-01-31:
Works for me.

whispercoros, 2017-04-16:
I have finally completed all of it, very glad to cross this one off the list...

closed 2017-04-16 by whispercoros