uf-mil/SubjuGator #195: Single thruster is not detected on first launch

duggerd, 2017-01-26:
Thruster with NID of 1 is not detected on the first launch of the sub from reboot. Relaunching the sub makes the thruster appear, and is detected on every subsequent launch until another reboot.

mattlangford, 2017-01-26:
Maybe we could try resetting the ID to 8 like we tried briefly for the other thruster and seeing if it'll come up right away

whispercoros, 2017-01-31:
Was this resolved at a past pool day?

duggerd, 2017-01-31:
No, the problem still exists. Have not tested changing the ID yet.

DSsoto, 2017-04-03:
Currently trying to get thruster coms and diagnosis a little more stable but no we never figured out why.

duggerd, 2017-04-04:
Let's write a couple lines into the thruster driver to log what was recieved and transmitted to a file so we can see what's happening, That's probably easier than trying to monitor a serial port directly.

Looks like only thruster ID 0 is returning any confirmation after a thrust command is sent. All the other thrusters stay silent but process and produce the thrust. All the thrusters are running identical firmware, so this is very strange.

duggerd, 2017-04-07:
It does not look like we have problems detecting thruster ID 0 on the first launch anymore. Also confirmed that ID 0 is a valid ID for a thruster based on the VideoRay protocol reference. The VideoRay test tool still fails to get information for all thrusters except ID 0, however the ROS driver is able to get that data on all the thrusters. The commands sent from ROS and the test tool are similar, so this is still an open issue to be resolved.

PR #233 (2017-05-07): THRUSTERS: fix timing dependent thruster detection
### Thruster Comms
* Increase the serial port timeout from 0.01 s to 0.1 s.
* Parse the response to poll packets to confirm that the response packet
  is actually from the thruster we believe it to be.
* Instead of checking for about 100 thruster ID's on each port, only check the
  ones that have been declared on the thruster layout.
### ThrusterSpinner
* Changed allowable effort range to [-1, 1] so we can spin in both directions
* Added a thruster discovery mode to the thruster spinner utility so that we can
  find out what thrusters are plug [... cut]
closed 2017-05-07 by ironmig