uf-mil/SubjuGator #196: Investigate if magnetometer calibration is necessary after thruster swaps

duggerd, 2017-01-26:
Are the manufacturing differences in each new thruster enough to affect the magnetometer calibration?

whispercoros, 2017-01-31:
Were the new thrusters in use at the last pool day? The sub was very stable near the end.

duggerd, 2017-01-31:
Through testing it does not look like anything is being severely affected. There might be some small changes that we can not see through observation.

Here is some notes on it from uf-mil-archive: https://github.com/uf-mil-archive/getting-started/blob/master/magnetic-calibration.md

@forrestv Can you comment on this (you wrote that document)?

whispercoros, 2017-01-31:
Looking at the text, it appears that it is able to also account for the metal in the sub. Perhaps we should wait until the new chassis is built and assembled to do a full calibration.

duggerd, 2017-02-01:
Looks like only hardsoft calibrations are being applied, per https://github.com/uf-mil/Sub8/blob/master/command/sub8_launch/launch/subsystems/nav_box.launch

mattlangford, 2017-02-01:
According to that launch it looks like the dynamic is being applied as well
from imu -- `/imu/mag_raw` --> [dynamic compensation] -- `/imu/mag_lessraw` --> [hardsoft calibration] -- `/imu/mag` --> UKF

duggerd, 2017-02-01:
You're correct - I don't have a good understanding of launch files, saw the remap and just assumed it did that without changing anything.

forrestv, 2017-02-01:
The dynamic compensation may be running, but AFAIK it's not doing anything since the thruster drivers don't publish their magnetic field contributions like the Sub7 thruster drivers did. It's safe to remove from the launch file (though you'd have to keep everything connected by changing the remaps).

So, no, dynamic compensation is irrelevant.

The hard/soft calibration should still be done any time anything near the magnetometer changes and ideally whenever anything metal anywhere on the sub changes.

mattlangford, 2017-02-01:
Awesome, thanks!

closed 2017-02-01 by mattlangford