uf-mil/SubjuGator #198: Semaphore woes

whispercoros, 2017-02-02:
Upon trying to make Semaphore compatible with the new stack, I found that it only supports building on Ubuntu 14.04. Since we decided to make the full switch to Ubuntu 16.04, this is likely going to cause problems. As I see it now, there are a few options available to us:
* Keep using Ubuntu 14.04 with ROS Indigo for builds
    * There are differences between the stacks
    * They may be close enough for our purposes
    * That said, I do not recommend this
* Find another online CI platform
    * I have not done much research yet, but I plan to soon
* Roll our own
    * Namely, [Jenkins](https://jenkins.io/index.html) looks promising
    * We wouldn't have any limits on builds per month, processing power,build threads, etc.
    * We would have more control over the build process (e.g. each build would not be pulling dependencies for 15 min.)
    * It would be much simpler to integrate this with in-house, GPU accelerated, Monte Carlo testing on Gazebo if we wanted to down the road

I'm leaving this up to the group because whichever we decide on will have a pretty significant impact on the project, so fell free to debate.

mattlangford, 2017-02-02:
Since you'd be the main one heading that, how difficult would it be to get something like Jenkins up and running?

whispercoros, 2017-02-02:
I'm not entirely sure at the moment, but probably no longer than a week to get it functioning enough for basic building. I will do some more research when I have a bit of spare time. In the meantime, I might push some duct tape to the install script on Sub8 to make Semaphore work.

ironmig, 2017-02-02:
The standard for open source projects no adays is Travis CI https://travis-ci.org . They say it's free for open source projects and supports 16.04 vm's. I say we try Travis out of it's easy. We want our CI to be rock solid, and deploying our own might make it less reliable.

whispercoros, 2017-02-09:
@ironmig attempted to get Travis CI working, but they are also stuck in the 14.04 dark ages. The duct tape appears to be working on Semaphore, but there is an error when running the tests for ros_alarms. I am planning to start building the infrastructure for Jenkins this weekend, so speak now or forever hold your peace.

DSsoto, 2017-02-11:
I'm in favor of rolling our own. No objections here.

DSsoto, 2017-04-03:
@whispercoros how is the CI work coming along? ETA?

whispercoros, 2017-04-04:
I had just gotten the SSL certificate from @duggerd before the IEEE competition, so I will be setting up Apache tonight barring any unforeseen things I have to take care of. I also set up authentication with AD with some help from @duggerd, so you should all be able to log in with your MIL credentials. The only things left to do are get Docker set up to run our tests and integrate with Github. I will be using SubjuGator as my guinea pig for testing out Docker, so it will be the first repository we have building. I will need write access to the organization to set up the build hooks.

DSsoto, 2017-04-04:
Ok I'll get you write access to the repo while you work on this. Still not convinced there should be more than 2 or 3 that are able to push in the long run.

whispercoros, 2017-04-13:
While there are still some things that need to be done with Jenkins, the automated build pipelines appear to be working well.

closed 2017-04-13 by whispercoros