uf-mil/SubjuGator #201: Refactor / Improve blueview driver

ironmig, 2017-03-22:
I'm working on refactoring Tess's blueview sonar driver from #197. Hope to move to using the C++ library, cleaning up the code, and outputting something besides an image (pointcloud??)

@DSsoto I call dibs

RustyBamboo, 2017-03-22:
Jason and I had some discussion of using this class: http://www.blueview.com/documentation/ProViewer-SDK-4.0-Manual/classBVTSDK_1_1RangeProfile.html

DSsoto, 2017-03-23:
@ironmig your dibs will be honored. You don't have the information to generate a point cloud from a single frame, @kingkevlar will work on generating a 2d occupancy grid and later on someone could work on reconstructing a 3d cloud if they choose to, although this is no trivial task. It would be nice if you published ping data or a range profile besides just the image.

mattlangford, 2017-03-23:
You could do a 2d point cloud no? With a `laserscan` message

DSsoto, 2017-03-23:
Ehhh, you're right, I reserve the word point **cloud** for higher dimensional things.

mattlangford, 2017-03-23:
It's a manifold of your "higher" dimensional space

whispercoros, 2017-03-24:
Make sure that you pull request your changes to mil_common and use the domain name for the sonar instead of it's IP address. Also, your definitions are fake news @DSsoto.

DSsoto, 2017-04-03:
This should really be in the mil_common repo.
 Is this ready to close? Or will @kingkevlar be adding her stuff to the driver.

closed 2017-04-03 by ironmig