uf-mil/SubjuGator #205: Remove packages migrated to mil_common

DSsoto, 2017-04-03:
We should start removing duplicate packages in the repo.

This is some of what needs to go:
+ sub8_perception/*/sub8_vision_lib
+ sub8_sonar
+ hydrophones
+ paulboard
+ some of sub8_tools
+ some of sub8_msgs

Be sure to reference your PR if you do any of this and/or add to this list.

whispercoros, 2017-04-04:
Is taking care of this on NaviGator right now out of the question? I feel like if we don't it will be a mess to do next year.

whispercoros, 2017-04-04:
Also, I would advise that we merge [mil_common #15](https://github.com/uf-mil/mil_common/pull/15) before this.

DSsoto, 2017-04-04:
Have you even reviewed any of it or commented that it looks good?
No one has.

ironmig, 2017-04-09:
I'm going to make this my pet project for the week. Plan is as follows:

1. [Use git grep with awk](http://blog.jasonmeridth.com/posts/use-git-grep-to-replace-strings-in-files-in-your-git-repository/) to replace strings like 'sub8_ros_tools' with 'mil_ros_tools'
2. Look at diff to verify nothing crazy happened
3. Start testing things to see what's broken

whispercoros, 2017-04-13:
Can this be closed now that #216 has been merged?

closed 2017-04-13 by DSsoto