uf-mil/SubjuGator #208: Self Check Improvements

duggerd, 2017-04-03:
- Add an indication of which thrusters are not detected.
- Check lsusb to detect if the cameras are plugged in vs the cameras launch not running.
- Detect peripheral devices like the actuator box and the passive sonar (if they respond to a RS485 ping).

duggerd, 2017-04-04:
That's a good idea. Would save time when we have problems or have to change out a thruster.

whispercoros, 2017-04-04:
It would be fairly straight forward to write an RQT plugin for this.

duggerd, 2017-04-05:
What's the advantage of a RQT plugin over just a text based tool?

whispercoros, 2017-04-05:
I didn't realize this was just a request to the existing system, sorry. It should be pretty simple to set ROS parameters for the path to each device in /dev and check if the file exists from Python.

whispercoros, 2017-04-09:
@DSsoto suggested creating alarms so that we are notified of any changes.

DSsoto, 2017-04-16:
I addressed the 2 first points @duggerd  made and also created the thruster_spin_test utility, Some of these diagnostic tests should be reported to the alarm server though, I will put that on the back burner.

BTW, should we make the self_check run automatically when we launch the sub? Thoughts?

whispercoros, 2017-04-16:
As long as the sub is still in a totally inert and killed state for the test, this is fine. I don't want Ken getting any thruster surprises if someone forgets that this is a default launch behavior.

DSsoto, 2017-05-15:
We should also add to the self check:
- [ ] Check fps for all camera topics using rostopic hz (like @ironmig did for #250)
- [ ] Checking for degenerate thruster configurations (#252)