uf-mil/SubjuGator #230: Slow Camera Frame Rate

duggerd, 2017-05-03:
All cameras are running slow. Fastest framerate for the front cameras is 14 fps and down is 7 fps. Changing the framerate to auto or above 7 fps for the down camera in flycap causes a blank image in ROS and PointGrey unexpected ethernet packet length issues.

If I set the framerate in flycap to 30 fps or auto for the down camera and then reboot the camera and check the framerate, it stays at 30 fps. However when ROS connects to the camera and then disconnects, the fps setting reverts to 7 fps. Is that being set somewhere in ROS?

ironmig, 2017-06-20:
Apparently the version of the flycap SDK bundled by default (2.9) with the driver we're using might not be compatible with 16.04 ([source](https://github.com/ros-drivers/pointgrey_camera_driver/pull/73)). I downloaded the latest (2.11) Flycap SDK and compiled with driver from source with this version. 

Now when we set the frame rate setting in the driver is seems to listen. We've had the two front cameras running at 30fps and the down camera running at 15fps at the last test day with no connection problems. Things do get messed up if you attempt to run two instances of the driver at once, but otherwise we've been unable to break it with the new SDK.

whispercoros, 2017-07-18:
Is this fixed now that we have moved over to the official SDK and are using the `pointgrey_camera_driver` repository?

ironmig, 2017-07-18:
Though we're still not getting 30fps on the down camera, I think this is the best we can do and is fast enough for our perception programs.

closed 2017-07-18 by ironmig