uf-mil/SubjuGator #237: Fix thruster rostest

ironmig, 2017-05-10:
Since #233 was merged, the ```test_sub8_mapper``` will occasionally fail (likely random/timing depenent) in [some builds](https://ci.mil.ufl.edu/jenkins/blue/organizations/jenkins/uf-mil%2FSubjuGator/detail/PR-228/11/pipeline). Someone should look into that...

I'll update this with more details as the story unfolds.

DSsoto, 2017-05-23:
Was this fixed? I've stopped noticing CI build fails due to this.

whispercoros, 2017-05-24:
It seems to be very sporadic. The variance day to day seems to be based on server load (maybe...?). This issue only cropped up after about a month of using the CI and went away in a few weeks without any changes being made to the test, so it is likely a timing issue with the test as @ironmig stated.