uf-mil/SubjuGator #238: Increase window size for bus-voltage kill

DSsoto, 2017-05-11:
Right now, the bus_voltage kill triggers unnecesarily due to the amount of fluctuation in the bus voltage readings.

On our small pool outing on May 10, I noticed that when station-keeping in still waters, the bus voltage readings had a total spread of about 0.5 volts in a span of just about 2 seconds.

Although I did not record numbers, I would venture to say that the average spread during move commands and other transient thruster behaviour is about 2 volts.

I propose that we only kill the sub when the average bus voltage drops below the pre-defined limits for a window of about 30 seconds.

I will submit a PR doing this soon, but open to hearing other ideas/counterarguments.

mattlangford, 2017-05-11:
Kalman filter bus voltage!

DSsoto, 2017-05-11:
There isn't one bus voltage, all of the bus voltage packets from different thrusters spam to the same /bus_voltage topic. I doubt we can assume measurements with Gaussian error of an average bus voltage in this scenario. **_Ad-hoc!!_**      (**_tldr_**: overkill)

DSsoto, 2017-05-23:
Addressed by #247. The window size is currently 30 seconds but can be adjusted based on observations from the next pool day.

closed 2017-05-23 by DSsoto