uf-mil/SubjuGator #240: Get mission props Cadded up and spwaned in sim

DSsoto, 2017-05-11:
We need to get all of the mission props spawned into the sim.

I'll leave a list here so that everyone is aware and can note progress on these. Leave a note if you work on this and update the date.

### Currently Spawned in `duck.launch`
- [ ] 
- [ ] 
- [ ] 
- [ ] 
- [ ]

whispercoros, 2017-05-13:
@Grymestone is taking this one over. He will be starting with a basic sandbox and working his way up to being able to dynamically generate a course of an arbitrary number of challenges.

ironmig, 2017-05-15:
I put all of Cameron's models on the file server under ```Sim Field Elements/2017```. The colors did not export for solid works so someone has to do the blender thing and color them or something.

DSsoto, 2017-05-23:
@Grymestone I know you've been doing good work on this. Make sure you document the progress here.