uf-mil/SubjuGator #253: Make path_marker_finder generic

DSsoto, 2017-05-15:
The competition requires finding the pose of multiple bright orange polygons objects repeatedly.

Kevin has implemented this successfully for the path marker but there is no reason to not use it for the bin lids as well.

I propose changing the name of the perception code from `path_marker_finder` to `orange_rectangle_finder` or some other more generic name. It should have a /set_geometry service that allows you to set the width and height of the object you want to find the pose of.

whispercoros, 2017-05-15:
I can look in to this if @ironmig doesn't want to since I will be handling this part of the perception for the bins challenge anyway.

ironmig, 2017-05-15:
I got this. Already doing that so it can work with unit tests

closed 2017-05-23 by ironmig