uf-mil/SubjuGator #254: Publish a persistent ogrid

DSsoto, 2017-05-15:
Right now @RustyBamboo's Ogrid generation works really well for a single frame of Imaging Sonar data, but it will forget about occupied and unoccupied regions as soon as they go out of the field of view.

An OGrid should should be generated that remembers the occupancy status of regions that we have previously observed based on a probabilistic model.

whispercoros, 2017-05-15:
Since some of our competition elements will be swaying with the currents, we need to look at how much object dilation we will experience. Unless of course that is part of the probabilistic model.

LucasBA, 2017-05-16:
I remember the competition elements being fairly stationary at the transdec. I think the current in there is pretty much constant and stuff really only moves around when we bump into it. 

@RustyBamboo are you working on this right now? If not I am interested in working on it today.

RustyBamboo, 2017-05-16:
@LucasBA I'm not currently working on it. Feel free to work on it