uf-mil/SubjuGator #260: Dump state of alarm server to file on kill

DSsoto, 2017-05-21:
To finally address issue #206:
* A dump alarms service should be added to the alarm server that returns the current state of all alarms on the system.
* The output from this service should be saved to a file in sub8_diagnostics after every kill.
* An alias should be added to quickly cat the file.

whispercoros, 2017-05-21:
Is this a SubjuGator specific issue, or should it just be added to ros_alarms in general?

DSsoto, 2017-05-23:
I think the fact that we're going to add behavior to the kill-handler on the sub makes it sub specific.

DSsoto, 2017-06-14:
@ironmig why was this closed? Did you implement this?

ironmig, 2017-06-14:
Oops thought this was the kill bag one

DSsoto, 2017-06-15:
No problem, you should always reference a PR or leave a comment when closing an issue though.