uf-mil/SubjuGator #270: Add checkpoints to Gazebo

Grymestone, 2017-05-23:
We need to add locations in the sim that when prompted in the console can teleport the sub in front of each objective. Currently, we have to maneuver to each objective prop. 

- [ ] Nav Gate Waypoint
- [ ] Start Gate Waypoint
- [ ] Bin Waypoint
- [ ] Octogon Waypoint
- [ ] Torpedo Board Waypoint
- [ ] Buoy Waypoint

DSsoto, 2017-05-23:
I'd recommend having a yaml under sub8_gazebbo/config with named world frame poses. Keep in mind that there is a difference between the map frame and  the world frame (the origin of the map frame is dependent on where the sub is spawned while the world frame is independent).

This will be great for integrating gazebo tests into CI.

whispercoros, 2017-05-24:
This is definitely a good idea, but I would recommend also looking into a way to generate custom worlds. For example, place a start gate at x1, y1, z1, p1, r1, y1 and the sub at x2, y2, z2, p2, r2, y2. This will be ideal for use with both a Monte Carlo testing engine as well as user-defined unit tests.

DSsoto, 2017-05-25:
Sure, but let's do things incrementally. I think this is something he could get done relatively quickly.

DSsoto, 2017-06-26:
@Grymestone addressed everything in #276.

closed 2017-06-26 by DSsoto