uf-mil/SubjuGator #277: Harder,  Better,  Faster, Better  pointcloud

RustyBamboo, 2017-06-07:
- Publish the conversion from blueview to pointcloud as a plain
- Make ogrid centered on Sub instead of 0,0 
- Rejects points below the DVL range
- Use the plainer pointcloud to create an action server that will move sub and return received pointcloud... This would allow missions or other nodes to do a "search", get a corresponding pointcloud, and then do some segmentation to find buoys or get poses.

whispercoros, 2017-06-08:
Not sure if this is in the same vein as point 4, but it would be nice to automate the filling in of unknown sections of the point cloud that we are trying to move into. This could be just a simple thing used for mission running (e.g. if the mission wants to move backward, spin the sub around to populate the occupancy grid, then spin it back and complete the move if that space is not occupied).

LucasBA, 2017-06-10:
@whispercoros I think this should be a separate thing like a conservative obstacle avoidance movement mode because I can think of scenarios where in a mission you would want to strafe or go backwards while keeping something in the view of the camera/sonar so no spinning.

LucasBA, 2017-06-10:
I am implementing point 4 to mirror the vision proxy in sub_singleton.