uf-mil/SubjuGator #288: Simulator is too dark

ironmig, 2017-06-24:
The lighting in gazebo is really dark, which makes it difficult to use for perception testing. For navigator, we had the sim very bright, which allowed us to test in ideal conditions. We should move the light, add new ones, or make the lights brighter so everything is easily visible

ironmig, 2017-06-24:
For example, this is an actual screenshot from the simulated down camera:
[image]

Grymestone, 2017-06-26:
I added a couple of light points and noticed its still pretty dark in some shots. Looking at it, there are spots that are exceedingly dark in the transdeck texture. It might be worth swapping out textures for something lighter. This is a screenshot of the ellipse table as seen from the down camera. By the bins, I still get a black screen essentially as with yours.
[image]

whispercoros, 2017-07-12:
Don't know if this has been fixed yet, but have a look at my lighting configuration from the NaviGator simulator. Specifically the [world file](https://github.com/uf-mil/NaviGator/blob/master/simulation/navigator_gazebo/worlds/a_whole_new.world). Basically, I just created 6 directional lights. Since most of the competition elements were vertical or horizontal planes, this added a lot of light to the simulator.

PR #390 (2019-02-22): GAZEBO: overhaul to simulate state sensors / dynamics
closes #387 
closes #388 
closes #372 
closes #327 
closes #288 

Significant overhaul of the gazebo simulation.
Features:

- simulation of dvl
- simulation of imu
- simulation of magnetometer
- simulation of depth sensor
- simulation of sub8's dynamics (drag, buoyancy, gravity, inertials)
closed 2019-02-22 by ironmig