uf-mil/SubjuGator #289: 'Fake' Simulator Perception

Grymestone, 2017-06-26:
In the event a mission's code is finished but the perception required for the mission is not, it would be a good idea to have a 'fake' perception service that can act as a substitute so missions can be tested. This service would return the correct coordinates of the object in Gazebo and fulfill all the requirements of regular perception. 

The idea is to make this pretty modular, all the mission writer would need to do to get this working is add a single word to the fake perception file adding their service to the list. I might also just iterate through the vision proxies file and get the services from there. Then the mission writer wouldn't need to do anything other than launch the service when they wanted to test their mission in gazebo without real perception. That would be ideal I imagine. All other values would be pulled from the launch file.

ironmig, 2017-07-07:
#295

DSsoto, 2017-07-15:
#295 was merged. Good work @Grymestone

closed 2017-07-15 by DSsoto