uf-mil/SubjuGator #318: Question about the Thrust Allocation Matrix Setup On Wiki

atomoclast, 2017-08-15:
Hello,
I was reading through your wiki located here: https://github.com/uf-mil/SubjuGator/wiki/Thrusters about the thruster matrix B, and how you derive it and I was wondering if you could explain what you mean by position vector and direction vector. 

Is the position vector [x,y,z] translation of the thruster from the root frame? Is the direction vector the vector that describes the direction of thruster force? So if you have one thruster pointing in the negative X direction, the direction vector is then [-1,0,0]?

Thanks in advance,

Andrew

RustyBamboo, 2017-08-15:
Hello.
Yes, as you describe, the position vector is the translation relative to the center of geometry (not center of mass) of the sub. And, yes, the direction vector describes the direction of the thruster, and your example is correct. 
The sub's horizontal thrusters are at 30-degree inwards, meaning the direction vector is a positive/negative permutation of [0.866, 0.5] with no z-competent giving [0.866, 0.5, 0.0]. Our four vertical thrusters point upwards, giving [0.0, 0.0, 1.0] for each of them.

This is the config file: https://github.com/uf-mil/SubjuGator/blob/master/gnc/sub8_thruster_mapper/config/thruster_layout.yaml

jpanikulam, 2017-08-17:
How on earth did you find this?

Q: How do you derive this

A: Derive this by constructing a matrix mapping magnitude of force at each thruster to force and torque about the origin of the frame.

Think of each row of the top `3x8` block as mapping all of the thrusts into one component of the net force. Think of each row in the bottom `3x8` block as mapping all of the thrusts into one component of the net torque about the origin. I encourage you to attempt this exercise at home.

Also of note:

* If the origin is offset from the center of rotation by some error `e`, then the estimated torque differ from that about the center of rotation as as `torque_about_center_of_rotation = the_torque_you_just_computed + (net_force x e)`. You can gauge your future sadness by the size of `e`.

* The page is wrong about the note regarding norm(u). If you know the mapping between u and power, it will absolutely change the optimal thrust configuration, even if monotonic.

* Now that I have said this, people with actual knowledge of mechanics will tell us how I am wrong.

closed 2019-01-08 by ironmig