uf-mil/SubjuGator #369: Implement Adaptive Controller

ironmig, 2019-01-05:
Implement an adaptive controller, similar to the [one on NaviGator](https://github.com/uf-mil/NaviGator/blob/master/gnc/navigator_controller/nodes/mrac_controller.py).

RustyBamboo, 2019-01-06:
https://github.com/RustyBamboo/ReeControl

ironmig, 2019-01-08:
What's your current state of the project and what are your near term plans for it? From skimming it it looks like a PID controller simulator and a basic 3D kinematic simulator with linear drag.

RustyBamboo, 2019-01-08:
I have the math figured out, just need to code it up and test it

ironmig, 2019-01-08:
"It's just code".  - Ira Hill

The README just describes a simulator... are you planning on also implementing an adaptive controller?

RustyBamboo, 2019-01-08:
yeah, I've been testing controllers in that simulator